horihiro/esp8266-google-home-notifier #2: ESP8266mDNS.h declaration before esp8266-google-home-notifier.h causes compilation errors

debsahu, 2018-07-08:
Using git version of arduino-esp8266, Arduino IDE 1.8.5, Windows 10 x64.

Adding `#include <ESP8266mDNS.h>` before `#include <esp8266-google-home-notifier.h>` causes compilation issues. I'm guessing the internal ESP8266mDNS.h is interfering with this.

```
C:\Users\xxxx\Dropbox\Arduino\libraries\arduino_244670\src\esp8266-google-home-notifier.cpp: In member function 'boolean GoogleHomeNotifier::device(const char*, const char*, int)':

C:\Users\xxxx\Dropbox\Arduino\libraries\arduino_244670\src\esp8266-google-home-notifier.cpp:44:33: error: 'class MDNSResponder' has no member named 'txt'

           if (strcmp(name, MDNS.txt(i, "fn").c_str()) == 0) {

                                 ^

Multiple libraries were found for "ESP8266mDNS.h"
 Used: C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS
 Not used: C:\Users\xxxx\Dropbox\Arduino\libraries\arduino_244670
```

horihiro, 2018-07-24:
Thanks @debsahu and sorry for late reply.

This library declares `MDNS` in internal `ESP8266mDNS.h`.
So you need to add `#include <esp8266-google-home-notifier.h>' before `#include <ESP8266mDNS.h>` and some header files which includes it.

This internal `ESP8266mDNS.h` will be included in [Arduino core for ESP8266 WiFi chip 2.5.0](https://github.com/esp8266/Arduino/pull/3107), then I will remove the internal one from this library.

debsahu, 2018-07-24:
I understand, thanks for the awesome library!

debsahu, 2018-07-24:
I heard that google is going to monetize translation service, dont know how it will effect tinkerers like us!

closed 2018-07-24 by debsahu

arreisau, 2018-12-30:
Hi, I followed the trail to here because I was getting the same or related compile errors.

In the setup instructions it says:
"use Arduino Core for ESP8266 2.5.0, which will merge ESP8266mDNS.cpp/.h, or later."

However, that pull request doesn't appear to have been merged in as esp8266 completely changed the mDNS implemenation.

"Closing in favor of #5384 ."
which was then:
"closing in favour of #5442"

and that's what seems to have ended up in 2.5.0

I haven't quite sorted out where the txt() function ended up, if they included one.

JLNapo, 2019-02-16:
Not work with Arduino IDE 1.8.8.
Worked very well with older versions. Can GCC4.4 be the cause?

...\esp8266-google-home-notifier\src\ESP8266mDNS.cpp: In member function 'bool MDNSResponder::_listen()':
...\esp8266-google-home-notifier\src\ESP8266mDNS.cpp:300:47: error: no matching function for call to 'UdpContext::listen(const ip_addr_t&, const int&)'
     if (!_conn->listen(*IP_ADDR_ANY, MDNS_PORT)) {
                                                                            ^
...\esp8266-google-home-notifier\src\ESP8266mDNS.cpp:300:47: note: candidate is:

In file included from ...\esp8266-google-home-notifier\src\ESP8266mDNS.cpp:60:0:

...\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpContext.h:98:10: note: bool UdpContext::listen(const ip_addr_t*, uint16_t)
     bool listen(CONST ip_addr_t* addr, uint16_t port)
            ^
Etc...

horihiro, 2019-02-16:
Could you please try esp8266 core 2.4.x?
This library doesn't work with 2.5.0 or later.

Please refer to #14 .

And open new issue in order to notify me if you have problems yet.

JLNapo, 2019-02-16:
Hello,

I set up version 2.4.2 and the compilation goes smoothly.

I wanted to recompile my program because the phrases sent were no longer heard by googlehome.

I'm afraid Google is deleting its APIs as it has been saying lately.

I will try, and I'll let you know as soon as possible.

Thanks for your help.

Regards,

Jean-Luc