horihiro/esp8266-google-home-notifier #3: Doorbell button press

computerfreek274, 2018-07-18:
I am having have been at this for a few day's now. I can not see to get a pull down from pin D2 to work so I can have GHN speak there is someone at the door. If anyone can help that would be great. Thanks. Here is the code I have so far it seems to be way off.

#include <ESP8266WiFi.h>
#include <Bounce2.h>                      // https://github.com/thomasfredericks/Bounce2
#include <esp8266-google-home-notifier.h> // https://github.com/horihiro/esp8266-google-home-notifier
                                          // https://github.com/horihiro/esp8266-google-tts

#define BUTTON_PIN 4   // D2/GPIO4 

const char* ssid     = "<REPLASE_YOUR_WIFI_SSID>";
const char* password = "<REPLASE_YOUR_WIFI_PASSWORD>";

Bounce debouncer = Bounce();
GoogleHomeNotifier ghn;

void setup() {

  // Setup the button with an internal pull-up :
  pinMode(BUTTON_PIN,INPUT_PULLUP);

  // After setting up the button, setup the Bounce instance :
  debouncer.attach(BUTTON_PIN);
  debouncer.interval(5); // interval in ms

  Serial.begin(115200);
  Serial.println("");
  Serial.print("connecting to Wi-Fi");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(250);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("connected.");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());  //Print the local IP
  
  const char displayName[] = "Family Room";  /// Change this, must match the name of Google Home

  Serial.println("connecting to Google Home...");
  if (ghn.device(displayName, "en") != true) {
    Serial.println(ghn.getLastError());
    return;
  }
  Serial.print("found Google Home(");
  Serial.print(ghn.getIPAddress());
  Serial.print(":");
  Serial.print(ghn.getPort());
  Serial.println(")");
}

void loop() {
  debouncer.update();
  if ( debouncer.fell() ) {
    if (ghn.notify("Someone is at the door!") != true) {
      Serial.println(ghn.getLastError());
      return;
    }
  }
}

horihiro, 2018-07-24:
What you want to do is to notify to some google homes, isn't it?

This library doesn't support to notify to multiple google homes.
Maybe, drastic design change is needed to support it. 

\# I think you can do that if you prepare as many ESPs as Google Home and each Google Homes have different name.

computerfreek274, 2018-07-25:
What about if I just Define another GoogleHomeNotifier for each google home. I have it setup and wired to a doorbell now. I will try it that way if I can get the code right. So far it has been working great for a doorbell no need for ifttt or any other outside server. Just connected for my wifi and done.

horihiro, 2018-07-25:
Is what you want to know the way to notify to multiple google-homes concurrently?

Maybe it's not impossible by current internal design.

I don't have a plan to support it because I don't have such an environment(I have only one google-home).

horihiro, 2018-11-04:
could I close this issue?

horihiro, 2018-11-05:
close

horihiro, 2018-11-05:
close

closed 2018-11-05 by horihiro