horihiro/esp8266-google-home-notifier #4: Arduino-ESP32 のコンパイルエラー

mgo-tec, 2018-08-05:
mgo-tec と申します。

今日初めて使わせていただきましたが、コンパイルが通りませんでした。
ESP32-DevKitC と M5Stack でも同じエラーです。

`esp8266-google-home-notifier.cpp:28:41: error: format '%X' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Werror=format=]`

`esp8266-google-home-notifier.cpp:243:50: error: format '%d' expects argument of type 'int', but argument 3 has type 'long unsigned int' [-Werror=format=]`

とりあえず、以下のように修正してみたら、コンパイルが通って動作確認できました。
これが正しいのかは良く分かりません。

line 20:
`long unsigned int chipid;`

line 28:
`sprintf(hostString, "ESP_%06lX", chipid);`

line 243:
`sprintf(this->m_clientid, "client-%ld", millis());`

Google Home が自発的に喋ったのには感動しました。
素晴らしいです。
ありがとうございます。

horihiro, 2018-08-11:
対応しました。
リリースまではしばしお待ちください。

closed 2018-08-11 by horihiro

mgo-tec, 2018-08-13:
早速の対応、感謝いたします。
無事、Arduino IDE のコンパイラの警告を「全て」にしてもコンパイル通りました。
ありがとうございました。

commit e1ac85b8 (2018-08-11): resolve #4