horihiro/esp8266-google-home-notifier #5: ip 関数と play 関数についてのご相談

mkttanabe, 2018-08-09:
田辺と申します。6月に ESP32 への対応を完了されて以来、
esp8266-google-home-notifier を活用させて頂いております。
先日より二度に渡り貴ライブラリを題材に勤務先のブログにて
以下の記事を公開させて頂きました。宜しければお手すきの折にご笑覧下さい。
http://dsas.blog.klab.org/archives/2018-08/52297379.html
http://dsas.blog.klab.org/archives/2018-08/52297568.html

さて、手元では fork したライブラリコードをプライベートにカスタマイズして
扱っているのですが、もし今後改訂のご予定があれば、
オリジナルの google-home-notifier に存在する ip 関数と play 関数 (public) を
貴ライブラリへ加味して頂けないでしょうか？ 
デバイスの IP アドレスを固定可能な環境では ip 関数は有用ですし、
また、MP3 データのキャスト再生が可能であればさらに使途が広がるのではないかと
想像しております。
ご多忙のところ恐れ入りますが、おってご検討頂ければ幸いです。

horihiro, 2018-08-09:
既にフォークしてカスタマイズしているんですね！
該当部分だけでもPRしていただければ、マージしますがいかがでしょうか。

mkttanabe, 2018-08-10:
早々にお返事をありがとうございます。後ほどリクエストをお送りしますので、内容をご確認の上ご判断下さい。

horihiro, 2018-08-11:
マージさせていただきました (fa7fd50)。
その後細かな修正をしているので、リリースまではしばしお待ちください。

closed 2018-08-11 by horihiro

mkttanabe, 2018-08-11:
（早々にありがとうございました。後日のリリースを楽しみにお待ちします。

horihiro, 2018-08-13:
昨日、リリースしました！