horihiro/esp8266-google-home-notifier #11: Failed to Connect with ESP8266 Wemos D1 mini

katze, 2018-10-29:
Hi,
I am experiencing random connection problems between a Wemos D1 mini (Esp8266) and Google Home.
Most of time i've get this error : "Failed to Connect to 192.168.1.11:8009."

I've tried connecting with device name or ip, sending wav or TTS but it does not change anything.
I'm using the last version of ESP8266 (2.4.2) installed using Arduino IDE Boards Manager

Do you have any advice ?

horihiro, 2018-11-04:
Do you have any information about this issue?

katze, 2018-11-04:
Hello,
Sorry I was off for a few days.
Notifications works about one in ten times.
Other times I get the error : "Failed to Connect to 192.168.1.11:8009" after a long timeout.
Maybe WiFiClientSecure is the cause, but i can't get logs.

192.168.1.11 is of course the ip of my Google Home Mini.

horihiro, 2018-11-22:
hmm, I can't reproduce this issue.

horihiro, 2018-12-02:
close once.

closed 2018-12-02 by horihiro

AjJordy, 2019-03-18:
I get almost the same issue. But I my case, always failed to connect. 
`
connecting to Wi-Fi.........................
connected.
IP address: 192.168.1.126
connecting to Google Home...
found Google Home(192.168.1.163:8009)
Failed to Connect to 192.168.1.163:8009.
`

robomaniac, 2021-02-21:
I got it work on both the **Wemos D1 mini** and **Wemos D1 mini Pro**

only works with the **google nest hub** and not my regular **google nest mini**

I use this

`IPAddress ip(192, 168, 1, 205);`

```
void connectToGH(){
  Serial.println("connecting to Google Home...");

  if (ghn.ip(ip, "en", 8009) != true) {                           // this use the IP address
    Serial.println(ghn.getLastError());
    return;
  }

  Serial.print("found Google Home(");
  Serial.print(ghn.getIPAddress());
  Serial.print(":");
  Serial.print(ghn.getPort());
  Serial.println(")");
}
```