horihiro/esp8266-google-home-notifier #12: IP address 0.0.0.0:0

krupaw2, 2018-10-31:
Any suggestion?

connecting to Google Home...
found Google Home(0.0.0.0:0)
Google Home's IP address/port is not set. Call 'device' or 'ip' method before calling 'cast' method.

horihiro, 2018-11-01:
I haven't experienced that.

Could you please check your Google Home'a IP address using Google Home App for smartphone?

krupaw2, 2018-11-01:
My address is 192.168.0.100

I try add:
ghn.ip(192.168.0.100, "en", 5353);
but i recive compilation error:
GoogleSay:30:8: error: too many decimal points in number

 ghn.ip(192.168.0.100, "en", 5353);

horihiro, 2018-11-01:
The type of first argument of method 'ip' is IPAddress class.

Please refer to following.
https://www.arduino.cc/en/Reference/EthernetIPAddress

And Google Home uses port 8009 usually.
How did you decide the port number?

krupaw2, 2018-11-02:
Thank you very much for your commitment.
I found such a hint somewhere on YT

I changed to:
ghn.ip ((192, 168, 0, 100), "en", 8009);
Now it compiles correctly, but unfortunately it does not work - receives message:
found Google Home (100.0.0.0:8009)

When I try to send text through the website in the logs I have:
Failed to Connect is 100.0.0.0:8009.

horihiro, 2018-11-04:
could I close this issue?

krupaw2, 2018-11-04:
Of course - everything works perfectly :)

closed 2018-11-04 by krupaw2