horihiro/esp8266-google-home-notifier #13: mDNS timeout.

siamkater, 2019-01-10:
Hello,
i own 5 google home speaker and all up and running but
It does not matter what google home name I enter, I only get this error
 
connecting to Wi-Fi.
connected.
IP address: *92.*68.*88.*84
connecting to Google Home...
mDNS timeout.

can you help me ?

horihiro, 2019-01-11:
could you please let me know whether this issue is reproduced if only one google home is turned on?

and which do you use, esp8266 or esp32?

siamkater, 2019-01-11:
shame on me displayname is case sensitive after i changed this it works.
 BUT the next problem i have is the speaker plays the signal tone but the text is missing this happens all 2-3 times. in the log i get no error.

connecting to Google Home...
found Google Home
192.168.188.54
:
8009
)
Done.

and i dont know if this feature talk to a group should work
if yes then i have the following problem
one speaker plays the notificatinon sound and sometimes a other speaker plays the text 

connecting to Google Home...
found Google Home
192.168.188.61
:
42524
)
Done.

i use the esp8266

horihiro, 2019-01-11:
sorry , I'm not sure that this library can make multiple google homes speak because i have only one google home.

do your google homes share same display name?

siamkater, 2019-01-11:
no they don't share the same display name 
in the google home app you can add the speaker to a group with a new display name for the group

horihiro, 2019-01-12:
I'm not sure that speaker **group** is related to this issue.

If you know, could you please let me know the reproducing steps of this issue when there is one Google Home,?

horihiro, 2019-02-11:
I close this issue once.
Please open again when repro steps is discovered.

closed 2019-02-11 by horihiro