horihiro/esp8266-google-home-notifier #14: Please Help

saintchelo, 2019-02-11:
Hi

I´m using arduino IDE 1.8.8 and Esp8266 2.5 core

In your Library I deleted ESP8266mDNS.cpp and ESP8266mDNS.h

in your sample esp8266>NotificationServer I have these first 4 lines

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <esp8266-google-home-notifier.h>
#include <ESP8266mDNS.h>

When I try to compile to my board Nodemcu 1.0 I got this error:

**C:\Users\MarceloLG\Documents\ARDUINO-VARIOS\Arduino\libraries\esp8266-google-home-notifier\src\esp8266-google-home-notifier.cpp: In member function 'boolean GoogleHomeNotifier::device(const char*, const char*, int)':

C:\Users\MarceloLG\Documents\ARDUINO-VARIOS\Arduino\libraries\esp8266-google-home-notifier\src\esp8266-google-home-notifier.cpp:33:33: error: 'using MDNSResponder = class esp8266::MDNSImplementation::MDNSResponder' has no member named 'txt'

           if (strcmp(name, MDNS.txt(i, "fn").c_str()) == 0) {

                                 ^

exit status 1
Erro compilando para a placa NodeMCU 1.0 (ESP-12E Module)**

Can you help me please ?

Thanks so much

horihiro, 2019-02-11:
Sorry for your inconvinient.

I haven't made this library work using ESP8266 core 2.5 and later.
\# README.md was written before releasing 2.5, so I have to fix the document.

Could you please rollback it to 2.4.1, and copile again?

saintchelo, 2019-02-11:
> Sorry for your inconvinient.
> 
> I haven't tried ESP8266 core 2.5 with this library.
> # README.md was written before releasing 2.5, so I have to fix the document.
> 
> Could you please rollback it to 2.4.1, and copile again?

Thanks for your attention

I rollback to 2.4.1 and occurs the same error:

C:\Users\MarceloLG\Documents\ARDUINO-VARIOS\Arduino\libraries\esp8266-google-home-notifier\src\esp8266-google-home-notifier.cpp: In member function 'boolean GoogleHomeNotifier::device(const char*, const char*, int)':

C:\Users\MarceloLG\Documents\ARDUINO-VARIOS\Arduino\libraries\esp8266-google-home-notifier\src\esp8266-google-home-notifier.cpp:33:33: error: 'class MDNSResponder' has no member named 'txt'

           if (strcmp(name, MDNS.txt(i, "fn").c_str()) == 0) {

                                 ^

exit status 1
Erro compilando para a placa NodeMCU 1.0 (ESP-12E Module)

It would be nice if I can cast audio phrase from Nodemcu to my Google Home

Any Idea ?

Thanks again

horihiro, 2019-02-11:
You wrote :

```
In your Library I deleted ESP8266mDNS.cpp and ESP8266mDNS.h
```

Could you please let me know if the error occurs when the files aren't deleted?

saintchelo, 2019-02-11:
> You wrote :
> 
> ```
> In your Library I deleted ESP8266mDNS.cpp and ESP8266mDNS.h
> ```
> Could you please let me know if the error occurs when the files aren't deleted?

FANTASTIC !!!!!   

If I don't delete the files and WORKS GREAT !!!!!

Thank you, Thank you SO MUCH !!!!!

I can cast a audio phrase from my nodemcu to my Google Home.

Excelent Library. Very Good Job.

I have only one more question please:

This Lybrary works only if my Google Home and my nodemcu be  in the same local network.

But if I wat to cast audio from nodemcu to my Google Home over Internet, its possible ?

I guess that in this case will be necessary a Google api ? Or there is other way ?

THANK YOU VERY MUCH !!!!!

saintchelo, 2019-02-11:
And other question please:

I notice that the speech velocity in my google home when casting is a bit slow.

Ther is a way to increase the velocity of audio casted from nodemcu to Goggle Home ?

Thanks a lot 

By the way, this is a FANTASTIC Library !!!!

horihiro, 2019-02-11:
I'm glad to hear that you're succeeded to cast. :tada::tada:

I answer your two questions.

Question1: Is there any way to cast to Google Home from the internet?
-> No, it need to cast to Google Home directly, so something to cast should be in the local network which Google Home connected.

Question2: Is there any way to change speech velocity?
-> It's [esp8266-google-tts](https://github.com/horihiro/esp8266-google-tts)'s issue which this library depends on.
     esp8266-google-home-notifier library just cast url of mp3, which is generated by esp8266-google-tts, to Google Home.
     And I don't know how to change speech velocity.

saintchelo, 2019-02-12:
> I'm glad to hear that you're succeeded to cast. 
> 
> I answer your two questions.
> 
> Question1: Is there any way to cast to Google Home from the internet?
> -> No, it need to cast to Google Home directly, so something to cast should be in the local network which Google Home connected.
> 
> Question2: Is there any way to change speech velocity?
> -> It's [esp8266-google-tts](https://github.com/horihiro/esp8266-google-tts)'s issue which this library depends on.
> esp8266-google-home-notifier library just cast url of mp3, which is generated by esp8266-google-tts, to Google Home.
> And I don't know how to change speech velocity.

Thanks again.

I understood.

I need again a little help please:

In function ghn.notify("text to say"); I need to replace "text to say" by a variable.
I tried to create a String new_var = "hello"; but when I try to compile ghn.notify(new_var);
in arduino ide occurs the error:

no matching function for call to 'GoogleHomeNotifier::notify(String&)'

Can I doing something wrong ?

What kind of variable I can use to replace "text to say" in ghn.notify("text to say"); ?

Thanks so much for your attention again

horihiro, 2019-02-12:
It's simple C/C++ question.

You can't use a `String` variable as an argument of `notify` method.
You have to convert to `const char*`, like following:

```
 if (ghn.notify(phrase.c_str()) != true) {
```
https://github.com/horihiro/esp8266-google-home-notifier/blob/master/examples/esp8266/NotificationServer/NotificationServer.ino#L52

saintchelo, 2019-02-12:
> It's simple C/C++ question.
>
> You can't use a `String` variable as an argument of `notify` method.
> You have to convert to `const char*`, like following:
>
> ```
>  if (ghn.notify(phrase.c_str()) != true) [...]

THANKS SO MUCH !

Now all is working perfectly, like your PERFECT LIBRARY.

Thank you

closed 2019-02-12 by horihiro

JLNapo, 2019-02-16:
Hello,
I set up version 2.4.2 and the compilation goes smoothly.
I wanted to recompile my program because the phrases sent were no longer heard by googlehome.
I'm afraid Google is deleting its APIs as it has been saying lately.
I will try, and I'll let you know as soon as possible.
Thanks for your help.
Regards,
Jean-Luc

horihiro, 2019-03-29:
Hi @JLNapo 
Sorry for late reply because I don't watch closed issue.

I can't reproduce this.
Please create new issue if you still face this problem.

And please check using node.js with https://github.com/noelportugal/google-home-notifier before new issue.

Regards,