horihiro/esp8266-google-home-notifier #29: Error when compaling esp8266-google-home-notifier

JBX028, 2020-12-30:
Hi,
I copied/pasted the example and simply replaced the wifi info but the following error is raised when I compile. I installed the most recent versions of all libraries (esp8266, tts, notifier, etc..)
Any idea ?
Thanks in advance

Arduino: 1.8.13 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/johnnybaillargeaux/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/johnnybaillargeaux/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/johnnybaillargeaux/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_build_352144 -warnings=none -build-cache /var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_cache_826148 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /Users/johnnybaillargeaux/Documents/Arduino/GoogleHomeNotifier/GoogleHomeNotifier.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/johnnybaillargeaux/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/johnnybaillargeaux/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/johnnybaillargeaux/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_build_352144 -warnings=none -build-cache /var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_cache_826148 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /Users/johnnybaillargeaux/Documents/Arduino/GoogleHomeNotifier/GoogleHomeNotifier.ino
Using board 'generic' from platform in folder: /Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4
Using core 'esp8266' from platform in folder: /Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4
Detecting libraries used...
/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/include -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/lwip2/include -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_build_352144/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266 -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/variants/generic /var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_build_352144/sketch/GoogleHomeNotifier.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]
ResolveLibrary(ESP8266WiFi.h)
  -> candidates: [ESP8266WiFi@1.0]
/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/include -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/lwip2/include -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_build_352144/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266 -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/variants/generic -I/Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src /var/folders/2q/43bt0l057ndbqdxhstzzh8km0000gn/T/arduino_build_352144/sketch/GoogleHomeNotifier.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for ESP8266WebServer.h: [ESP8266WebServer@1.0]
ResolveLibrary(ESP8266WebServer.h)
  -> candidates: [ESP8266WebServer@1.0]
[... cut ...]
Using library ESP8266WebServer at version 1.0 in folder: /Users/johnnybaillargeaux/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WebServer 
Using library esp8266-google-home-notifier at version 1.0.7 in folder: /Users/johnnybaillargeaux/Documents/Arduino/libraries/esp8266-google-home-notifier 
Using library esp8266-google-tts at version 1.0.10 in folder: /Users/johnnybaillargeaux/Documents/Arduino/libraries/esp8266-google-tts 
exit status 1
Error compiling for board Generic ESP8266 Module.

JBX028, 2020-12-31:
Hi,
method setWIFIClientSecure comes from the TTS library (https://github.com/horihiro/esp8266-google-tts/blob/9a87b4627314c8e641ffcaeb03df7b33eb51008f/src/google-tts.cpp) which is however declared at the top of the program. 
Did I missed something ?

TheConsciousness, 2020-12-31:
> Hi,
> method setWIFIClientSecure comes from the TTS library (https://github.com/horihiro/esp8266-google-tts/blob/9a87b4627314c8e641ffcaeb03df7b33eb51008f/src/google-tts.cpp) which is however declared at the top of the program.
> Did I missed something ?

Looks like I'm in the same boat as you. Good timing!
Trying to compile this on my Wemos D1 mini board, but am getting:

`error: 'TTS' has no member named 'setWiFiClientSecure'
     tts.setWiFiClientSecure(m_client);
         ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
`

JBX028, 2021-01-01:
I am able to compile the esp8266-google-tts therefore I don't understand why this one cannot be.
To be honest, my esp8266 (nodeMCU model) is not recognised in my mac book pro. 
I have tried to follow this step https://medium.com/@bass.pj/nodemcu-esp8266-getting-started-with-arduino-on-macos-1cdc61565496 but without success so far.
Above saying, I think I will stop the investigations. Currently, the google notifier in JS works fine in my rasbperry PI but I wanted to use my PI for something else and take advantage of the nodeMCU via this library but it's hardest than expected unfortunately with the device not recognised and this error with the compilation (I don't think these 2 issues are linked).

horihiro, 2021-01-02:
Thank you for reporting.

This might be caused by updating [esp8266-google-tts to 1.0.10](https://github.com/horihiro/esp8266-google-tts/releases/tag/1.0.10).

I will fix this issue, but if you want to use this asap, please try to comment out [this line](https://github.com/horihiro/esp8266-google-home-notifier/blob/master/src/esp8266-google-home-notifier.cpp#L86) as following.

```c_cpp
//     tts.setWiFiClientSecure(m_client);
```