horihiro/esp8266-google-home-notifier #35: Can't push any notify to Google Home

rezaneam, 2021-03-07:
Hi @horihiro 

I just noticed that I can't push any notification to my Google Home Mini!
I receive the following code when triggering the notify method!
`[E][WiFiClientSecure.cpp:127] connect(): start_ssl_client: -1`
When I check the last error I get this
`Failed to Connect to 192.168.1.128:8009.`

It is a bit strange, the IP address is correct and this code previously worked on my setup.
I afraid if Google released a new update that changes the workflow.

System and cast firmware version of my Google Home are 229149 and 1.50.229.149

Can you please check if you have the same issue? or do you have access to any documentation on it?

Kind regards
Reza

horihiro, 2021-03-09:
This issue cannot be reproduced on my environment.

  - System and cast firmware version of Google Home: 229149 and 1.50.229.149 (same as you)
  - Board: ESP32 Dev Board (1.0.4 of [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32))
  - esp8266-google-home-notifier: 1.0.7

rezaneam, 2021-03-09:
@horihiro 
Thank you for your reply.  Last time I changed the Google Home speaker, WiFi router, firewall.
Finally, I found out it is [the platform](https://github.com/platformio/platform-espressif32) making problem. Probably in the last update, they broke something.

Thanks for your help.

rezaneam, 2021-04-17:
@horihiro
Hello again,
The issue I observed is still valid for [Ardunio ESP32](https://github.com/espressif/arduino-esp32) 1.0.5 and later, 
[After Ardunio 1.0.5](https://github.com/espressif/arduino-esp32/issues/4992#issuecomment-811088088) you need to call `m_client->setInsecure();` before calling `m_client->connect`. 
Can you please update the following part?

> #if defined(ARDUINO_ARCH_ESP8266) && !defined(ARDUINO_ESP8266_RELEASE_BEFORE_THAN_2_5_0)
>   m_client->setInsecure();
> #endif

Regards, Reza

brennomessana, 2021-04-21:
hello I used eso 1,.0,4 library  and with r32 azdelivery and tttgo Lora32 and work fine ,  but understand which line at modify in cpp source :   #if defined(ARDUINO_ARCH_ESP8266) && !defined(ARDUINO_ESP8266_RELEASE_BEFORE_THAN_2_5_0)
  m_client->setInsecure();
#endif
  if (!m_client->connect(this->m_ipaddress, this->m_port)) {
    sprintf(error, "Failed to Connect to %d.%d.%d.%d:%d.", this->m_ipaddress[0], this->m_ipaddress[1], this->m_ipaddress[2], this->m_ipaddress[3], this->m_port);
    this->setLastError(error);
    disconnect();
    return false;
  }

horihiro, 2021-04-23:
I already fixed and committed, will release soon.