horihiro/esp8266-google-home-notifier #36: Does notifier require a Chromecast device?

GaryDyr, 2021-03-24:
I have been unable to get notifier to speak through my Google Home Miini on my lan. The sketch finds the Mini, but always comes back with a "Failed to Connect to ........." message repeating the correct IP and port. Looking at the code, it seems to use the castMessage for the proto buffer. I am a rank neophyte in all this, but this seems to suggest that a Chromecast device must also be on the local lan, which I do not have. Is this true?

horihiro, 2021-03-25:
You don't need other Chromecast devices except for Google Home Mini.
Actually, I have only Google Home Mini, I don't have another Chromecast device.

brennomessana, 2021-04-01:
Hi the problem is't  google home mini but the board  device, if use other device like esp8266  esp12E Module the library work fine! , thanks a lot to HoriHiro for great job 

use this code for connect via IP  to google home nest mini 
i(s not mine )

GoogleHomeNotifier ghn;
void connectToGH() {
  IPAddress ip(192, 168, 1, 12);  // ip address of your google home mini nest 
 
 Serial.println("connecting to Google Home...");

  if (ghn.ip(ip, "IT", 8009) != true) {                           // this use the IP address and language italian 
    Serial.println(ghn.getLastError());
    return;
  }
}