horihiro/esp8266-google-home-notifier #37: Possible fix for ESP32 not working.

GaryDyr, 2021-04-22:
@Hirohito; @razaaneam; @brennonmessana
Hi,
I want to complement horihiro on what for me turns out to be a much needed valuable library. As a previous question suggested, I struggled and was failing to get an ESP32 to send a message to my Google Home Mini. I could see the call to Google translate was working, but like, others I was came up with "failed to connect to 1.92.168.1.3:8009. The ip address and port identified were correct.  Being a neophyte in understanding wired and wireless connections did not help. Based on comments by other posters. I think I finally fixed my issue, but the fix is different than their changes suggest.
Conditions:
Google Home Mini: Cast firmware 1.54.250118; System firmware version 250118,
ESP32: MeLife ESP32, Aruduino IDE Board: ESP32 Dev Module.
notifier tts version is 1.1.0 and notifier version: 1.07.

What I did is add another compiler directive as a test case. Starting at line 110 in esp8266-google-home-notifier.cpp. my code reads:

#if defined(ARDUINO_ARCH_ESP8266) && !defined(ARDUINO_ESP8266_RELEASE_BEFORE_THAN_2_5_0)
  m_client->setInsecure();
#endif
#ifdef ARDUINO_ARCH_ESP32
  m_client->setInsecure(); 
#else
#error "ARDUINO_ARCH_ESP32 change went weird."
#endif

The first 3 lines are part of the original code. With this change, I finally was able to get the Mini to announce my text message. Why declaring the ESP32 client as insecure is necessary, is beyond my current understanding, although I would certainly like to understand why.
As an aside: The desperation to find a solution stems from developing a simple sketch to create an alert when our house sump pump has a float that on rare occasions fails to trigger the pump, potentially leading to basement flooding. There is no way to create an audio alert through the Google Home Mini using Google Assistant or AWS Alexa that I was able to find.

GaryDyr, 2021-04-23:
@horilro
I downloaded and installed the latest commit for the notifier in my Arduino IDE library folder. Works fine for me. Thank you for the quick response.

closed 2021-04-23 by GaryDyr

horihiro, 2021-04-23:
Thanks for verifying, @GaryDyr