horihiro/esp8266-google-home-notifier #39: announcement does not work

ateker01, 2021-05-23:
Hello;

Couple days ago, all of a sudden my google home mini stopped doing the announcements. It only does the beep sound but would not announce "Hello World!"

I am using the example sketch (SimpleUsage) and below is the google home mini system information. Is this library still working for you? Maybe the Google translation services is not working?

System firmware version: 250118
Cast firmware version: 1.54.250118
IP Address: 192.168.1.79

Below is the console log from 8266

21:41:44.431 -> connecting to Google Home...
21:41:45.414 -> found Google Home(192.168.1.79:8009)
21:41:46.770 -> Done.

fingalo, 2021-05-26:
I have the same issue, just tested in ESP32. I can get the link of mp3 file using example in tts library
but notifier example just say 'plionk' on my mini.

horihiro, 2021-05-26:
Please verify if you can play the following MP3 URL by your browser and provide the versions for `esp8266-google-home-notifier` and `esp8266-google-tts`.

https://translate.google.com/translate_tts?ie=UTF-8&q=hello&tl=en&client=tw-ob&ttsspeed=1

ateker01, 2021-05-26:
I have the latest version of both of those libraries. 1.0.8 and 1.1.0 respectively.
Yes I can play the sound on my computer. the sound is not super clear but there is no drastic change.

horihiro, 2021-05-27:
I cannot repro this.
At least My ESP32 (it's hard to try 8266 now) can send a text message and Google Home can speak the message.

mark8018, 2021-05-27:
my environment

esp32: 1.0.6
Board: M5Stack-Core-ESP332
esp8266-google-home-notifier: 1.0.8
esp8266-google-tts: 1.1.0

GoogleHomeMini:
System firmware version: 250118
Cast firmware version: 1.54.250118

SimpleUsage:
connecting to Wi-Fi............
connected.
IP address: 192.168.11.45
connecting to Google Home...
found Google Home(192.168.11.14:8009)
Done.

pilo! (don't Speek Hello, World!)

Um....

tiagoalternativa, 2021-05-29:
has anyone found the solution to this problem?

Hak5-design, 2021-06-04:
Hello,

I have the same problem:  since some couple of days, all of a sudden my google home mini stopped doing the announcements. It only does the beep sound but would not announce "Hello World!" Does everyone have any solution?

Eutektikum, 2021-06-06:
Hello,

same problem here:

Board: NodeMCU
esp8266-google-home-notifier: 1.0.8
esp8266-google-tts: 1.1.0

GoogleHomeMini:
System firmware version: 250118
Cast firmware version: 1.54.250118

(exactly the same as mark8018)

Thanks in advance!

Eutektikum, 2021-06-08:
It works again. It might have been a problem on Google's side.

mark8018, 2021-06-08:
My Google home mini has also regained its words!

horihiro, 2021-06-10:
I close this because this issue has gone.

closed 2021-06-10 by horihiro