horihiro/esp8266-google-home-notifier #40: SEP 8266 Example Compile Error

pr-wiley, 2021-09-15:
When attempting to compile the "Simple Usage" example for 8266 I get this compile error:

/Users/prw/Documents/Arduino/libraries/esp8266-google-home-notifier/src/ESP8266mDNS.cpp: In member function 'MDNSTxt* MDNSResponder::_getServiceTxt(char*, char*)':
/Users/prw/Documents/Arduino/libraries/esp8266-google-home-notifier/src/ESP8266mDNS.cpp:569:42: error: cannot convert 'bool' to 'MDNSTxt*' in return
  569 |       if (servicePtr->_txts == 0) return false;
      |                                          ^~~~~
exit status 1
Error compiling for board LOLIN(WEMOS) D1 mini (clone).

DanBar1971, 2021-09-28:
I got the same error.

horihiro, 2021-09-28:
Sorry for the delay.
I confirmed that this issue happened on ESP8266 Arduino Core@3.x .

Please downgrade ESP8266 Arduino Core to 2.7.4 and try again if this is urgent though I will fix this when I can make time.

ddebasilio, 2022-02-03:
I'm also facing similar issues. I am unable to compile the latest version of the library 😢 .  Any chance that this could be fixed? I am building a weather station that should talk to google, but I would prefer to work with the latest core.

wiltwong, 2022-02-09:
Edit line 569 of ESP8266mDNS.cpp to return `nullptr` instead of `false`